busel7/JWSMurottal #3: Error compiling for board NodeMCU 1.0 (ESP-12E Module).

valkamel, 2021-01-05:
Arduino: 1.8.13 (Linux), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 460800"

In file included from /home/amanek/Arduino/Project/JWSMurottal/JWSMurottal.ino:23:0:
/home/amanek/Arduino/libraries/PrayerTimes/PrayerTimes.h:8:20: fatal error: String.h: No such file or directory
 #include <String.h>
                    ^
compilation terminated.
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

busel7, 2021-01-05:
try changing the file /home/amanek/Arduino/libraries/PrayerTimes/PrayerTimes.h

#include <String.h>

with

#include <string.h>
 
Default String.h with S uppercase change to string.h small s lowercase

busel7, 2021-02-13:
Mudah-mudahan sudah bisa, case ini kami close. Jika masih ada kendala bisa membuka case ini lagi, terimakasih.

closed 2021-02-13 by busel7