busel7/JWSMurottal #4: Error compiling for board NodeMCU 1.0 (ESP-12E Module).

alan268, 2021-01-05:
Assalamualaikum.,  mas Bonny.
Mohon pencerahannya.,  semoga berkah.

sketch\DFPlayer.h:24:15: note:   candidate expects 1 argument, 2 provided
In file included from D:\ABI ALAN\Arduino\JWS Mas Bonny\JWSmurotal200\JWSmurotal200.ino:9:0:
C:\Users\Aylin\Documents\Arduino\libraries\DFMiniMp3-master\src/DFMiniMp3.h:552:67: error: no matching function for call to 'Mp3Notify::OnError(DFMiniMp3<HardwareSerial, Mp3Notify>&, uint16_t&)'
                     T_NOTIFICATION_METHOD::OnError(*this, replyArg);
                                                                   ^
C:\Users\Aylin\Documents\Arduino\libraries\DFMiniMp3-master\src/DFMiniMp3.h:552:67: note: candidate is:
In file included from D:\ABI ALAN\Arduino\JWS Mas Bonny\JWSmurotal200\JWSmurotal200.ino:26:0:
sketch\DFPlayer.h:24:15: note: static void Mp3Notify::OnError(uint16_t)
   static void OnError(uint16_t errorCode)
               ^
sketch\DFPlayer.h:24:15: note:   candidate expects 1 argument, 2 provided
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

busel7, 2021-01-06:
Sudah coba yang terbaru? baru diupdate sekitar 3 hari lalu, perbaikan error saat compile terutama mengenai library dfplayer. Coba di download lagi sktect, jika masih ada error bisa diinfokan kembali.

busel7, 2021-01-11:
Coba install semua library yg diperlukan, linknya ada dideskripsi

busel7, 2021-02-13:
Mudah-mudahan sudah bisa, case ini kami close. Jika masih ada kendala bisa membuka case ini lagi, terimakasih.

closed 2021-02-13 by busel7