busel7/JWSMurottal #5: Error compiling for board NodeMCU 1.0 (ESP-12E Module).

alan268, 2021-01-06:
Assalamualaikum., Mas Bonny.
Sudah saya coba download ulang. hasinya masih sama.

Arduino: 1.8.13 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:1MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, All Flash Contents, 921600"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Aylin\AppData\Local\Arduino15\packages -hardware C:\Users\Aylin\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Aylin\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Aylin\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=legacy,ssl=all,eesz=4M1M,led=2,ip=lm2f,dbg=Disabled,lvl=None____,wipe=all,baud=921600 -ide-version=10813 -build-path C:\Users\Aylin\AppData\Local\Temp\arduino_build_969926 -warnings=none -build-cache C:\Users\Aylin\AppData\Local\Temp\arduino_cache_268195 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python3.path=C:\Users\Aylin\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Aylin\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Aylin\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\Aylin\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=C:\Users\Aylin\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=C:\Users\Aylin\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mkspiffs.path=C:\Users\Aylin\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\Aylin\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -verbose D:\ABI ALAN\Arduino\JWS Mas Bonny\JWSMurottal2021\JWSMurottal2021.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Aylin\AppData\Local\Arduino15\packages -hardware C:\Users\Aylin\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Aylin\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Aylin\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=legacy,ssl=all,eesz=4M1M,led=2,ip=lm2f,dbg=Disabled,lvl=None____,wipe=all,baud=921600 -ide-version=10813 -build-path C:\Users\Aylin\AppData\Local\Temp\arduino_build_969926 -warnings=none -build-cache C:\Users\Aylin\AppData\Local\Temp\arduino_cache_268195 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python3.path=C:\Users\Aylin\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Aylin\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Aylin\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\Aylin\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=C:\Users\Aylin\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=C:\Users\Aylin\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mkspiffs.path=C:\Users\Aylin\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\Aylin\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -verbose D:\ABI ALAN\Arduino\JWS Mas Bonny\JWSMurottal2021\JWSMurottal2021.ino

Using board 'nodemcuv2' from platform in folder: C:\Users\Aylin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4

Using core 'esp8266' from platform in folder: C:\Users\Aylin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4

Detecting libraries used...

"C:\\Users\\Aylin\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Aylin\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/include" "-IC:\\Users\\Aylin\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/lwip2/include" "-IC:\\Users\\Aylin\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\Aylin\\AppData\\Local\\Temp\\arduino_build_969926/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10813 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DLED_BUILTIN=2 -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\Aylin\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\cores\\esp8266" "-IC:\\Users\\Aylin\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\variants\\nodemcu" "C:\\Users\\Aylin\\AppData\\Local\\Temp\\arduino_build_969926\\sketch\\JWSMurottal2021.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for ESP8266WiFi.h: [ESP8266WiFi@1.0]

ResolveLibrary(ESP8266WiFi.h)

  -> candidates: [ESP8266WiFi@1.0]

"C:\\Users\\Aylin\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Aylin\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/include" "-IC:\\Users\\Aylin\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/lwip2/include" "-IC:\\Users\\Aylin\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\Aylin\\AppData\\Local\\Temp\\arduino_build_969926/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10813 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DLED_BUILTIN=2 -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\Aylin\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\cores\\esp8266" "-IC:\\Users\\Aylin\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\variants\\nodemcu" "-IC:\\Users\\Aylin\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\libraries\\ESP8266WiFi\\src" "C:\\Users\\Aylin\\AppData\\Local\\Temp\\arduino_build_969926\\sketch\\JWSMurottal2021.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for ESP8266WebServer.h: [ESP8266WebServer@1.0]

ResolveLibrary(ESP8266WebServer.h)

  -> candidates: [ESP8266WebServer@1.0]

"C:\\Users\\Aylin\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Aylin\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/include" "-IC:\\Users\\Aylin\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/lwip2/include" "-IC:\\Users\\Aylin\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\Aylin\\AppData\\Local\\Temp\\arduino_build_969926/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections  [... cut ...]
Using library HJS589 in folder: C:\Program Files (x86)\Arduino\libraries\HJS589 (legacy)

Using library SPI at version 1.0 in folder: C:\Users\Aylin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\SPI 

Using library Arduino-Prayer-Times-master in folder: C:\Users\Aylin\Documents\Arduino\libraries\Arduino-Prayer-Times-master (legacy)

Using library Hash at version 1.0 in folder: C:\Users\Aylin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\Hash 

exit status 1

Error compiling for board NodeMCU 1.0 (ESP-12E Module).

busel7, 2021-01-11:
Coba install semua library yg diperlukan, linknya ada dideskripsi

busel7, 2021-02-13:
Mudah-mudahan sudah bisa, case ini kami close. Jika masih ada kendala bisa membuka case ini lagi, terimakasih.

closed 2021-02-13 by busel7