busel7/JWSMurottal #6: WebSocketsServer error

abdghofur66, 2021-01-08:
Assalamu'alaikum Bang Bonny
sy punya permasalahan tentang websocket
mohon solusi, sebelumnya sudah sy update semua library, apa ada yang terlewat? Mohon solusi
[image]

In file included from D:\Download\Compressed\JWSMurottal\JWSMurottal.ino:31:0:

WebServer.h:61:1: error: 'WebSocketsServer' does not name a type

 WebSocketsServer webSocket = WebSocketsServer(81);

 ^

WebServer.h:63:34: error: 'WStype_t' has not been declared

 void webSocketEvent(uint8_t num, WStype_t type, uint8_t * payload, size_t length) {

                                  ^

sketch\WebServer.h: In function 'void webSocketEvent(uint8_t, int, uint8_t*, size_t)':

WebServer.h:65:14: error: 'WStype_TEXT' was not declared in this scope

   if(type == WStype_TEXT) {

              ^

D:\Download\Compressed\JWSMurottal\JWSMurottal.ino: In function 'void setup()':

JWSMurottal:68:3: error: 'webSocket' was not declared in this scope

   webSocket.begin();

   ^

D:\Download\Compressed\JWSMurottal\JWSMurottal.ino: In function 'void loop()':

JWSMurottal:88:3: error: 'webSocket' was not declared in this scope

   webSocket.loop();

   ^

exit status 1

'WebSocketsServer' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

busel7, 2021-01-11:
Coba install semua library yg diperlukan, linknya ada dideskripsi

busel7, 2021-02-13:
Mudah-mudahan sudah bisa, case ini kami close. Jika masih ada kendala bisa membuka case ini lagi, terimakasih.

closed 2021-02-13 by busel7