busel7/JWSMurottal #8: Error Compiling For Board ESP8266 Module

Pramfi, 2021-01-13:
Assalamualaikum Bang Bonny... sy sdh coba berkali kali hinga uninstall dan reinstall arduino mulai ver. 1.6xx ~ 1.8.13, semua library sdh upgrade dan downgrade tp hasilnya tetep sama " Error Compiling ..." dan sdh pake source code terbaru dari bang bonny. masalahnya error dimananya ya ?

busel7, 2021-01-13:
Waalaikumsalam, errornya apa? coba cek di Issue temen lainnya mungkin ada yg sama dan sudah dijawab.

Pramfi, 2021-01-13:
Error nya seperti ini Bang ...

Arduino: 1.8.13 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from D:\_BELAJAR JWS\NodeMCU_ESP8266\JWS_P5_RGB_64x32_Simple_ESP32-master\BONNY USEFUL\JWSMurottalMain\JWSMurottalMain.ino:9:0:

C:\Program Files\Arduino\libraries\DFMiniMp3-master\src/DFMiniMp3.h: In instantiation of 'uint16_t DFMiniMp3<T_SERIAL_METHOD, T_NOTIFICATION_METHOD>::listenForReply(uint8_t) [with T_SERIAL_METHOD = HardwareSerial; T_NOTIFICATION_METHOD = Mp3Notify; uint16_t = short unsigned int; uint8_t = unsigned char]':

C:\Program Files\Arduino\libraries\DFMiniMp3-master\src/DFMiniMp3.h:332:38:   required from 'uint16_t DFMiniMp3<T_SERIAL_METHOD, T_NOTIFICATION_METHOD>::getTotalTrackCount(DfMp3_PlaySource) [with T_SERIAL_METHOD = HardwareSerial; T_NOTIFICATION_METHOD = Mp3Notify; uint16_t = short unsigned int]'

sketch\DFPlayer.h:71:62:   required from here

C:\Program Files\Arduino\libraries\DFMiniMp3-master\src/DFMiniMp3.h:511:101: error: no matching function for call to 'Mp3Notify::OnPlayFinished(DFMiniMp3<HardwareSerial, Mp3Notify>&, DfMp3_PlaySources, uint16_t&)'

                         T_NOTIFICATION_METHOD::OnPlayFinished(*this, DfMp3_PlaySources_Usb, replyArg);

                                                                                                     ^

C:\Program Files\Arduino\libraries\DFMiniMp3-master\src/DFMiniMp3.h:511:101: note: candidate is:

In file included from D:\_BELAJAR JWS\NodeMCU_ESP8266\JWS_P5_RGB_64x32_Simple_ESP32-master\BONNY USEFUL\JWSMurottalMain\JWSMurottalMain.ino:26:0:

sketch\DFPlayer.h:31:15: note: static void Mp3Notify::OnPlayFinished(DfMp3_PlaySources, uint16_t)

   static void OnPlayFinished(DfMp3_PlaySources source, uint16_t track)

               ^

sketch\DFPlayer.h:31:15: note:   candidate expects 2 arguments, 3 provided

In file included from D:\_BELAJAR JWS\NodeMCU_ESP8266\JWS_P5_RGB_64x32_Simple_ESP32-master\BONNY USEFUL\JWSMurottalMain\JWSMurottalMain.ino:9:0:

C:\Program Files\Arduino\libraries\DFMiniMp3-master\src/DFMiniMp3.h:515:100: error: no matching function for call to 'Mp3Notify::OnPlayFinished(DFMiniMp3<HardwareSerial, Mp3Notify>&, DfMp3_PlaySources, uint16_t&)'

                         T_NOTIFICATION_METHOD::OnPlayFinished(*this, DfMp3_PlaySources_Sd, replyArg);

                                                                                                    ^

C:\Program Files\Arduino\libraries\DFMiniMp3-master\src/DFMiniMp3.h:515:100: note: candidate is:

In file included from D:\_BELAJAR JWS\NodeMCU_ESP8266\JWS_P5_RGB_64x32_Simple_ESP32-master\BONNY USEFUL\JWSMurottalMain\JWSMurottalMain.ino:26:0:

sketch\DFPlayer.h:31:15: note: static void Mp3Notify::OnPlayFinished(DfMp3_PlaySources, uint16_t)

   static void OnPlayFinished(DfMp3_PlaySources source, uint16_t track)

               ^

sketch\DFPlayer.h:31:15: note:   candidate expects 2 arguments, 3 provided

In file included from D:\_BELAJAR JWS\NodeMCU_ESP8266\JWS_P5_RGB_64x32_Simple_ESP32-master\BONNY USEFUL\JWSMurottalMain\JWSMurottalMain.ino:9:0:

C:\Program Files\Arduino\libraries\DFMiniMp3-master\src/DFMiniMp3.h:519:103: error: no matching function for call to 'Mp3Notify::OnPlayFinished(DFMiniMp3<HardwareSerial, Mp3Notify>&, DfMp3_PlaySources, uint16_t&)'

                         T_NOTIFICATION_METHOD::OnPlayFinished(*this, DfMp3_PlaySources_Flash, replyArg);

                                                                                                       ^

C:\Program Files\Arduino\libraries\DFMiniMp3-master\src/DFMiniMp3.h:519:103: note: candidate is:

In file included from D:\_BELAJAR JWS\NodeMCU_ESP8266\JWS_P5_RGB_64x32_Simple_ESP32-master\BONNY USEFUL\JWSMurottalMain\JWSMurottalMain.ino:26:0:

sketch\DFPlayer.h:31:15: note: static void Mp3Notify::OnPlayFinished(DfMp3_PlaySources, uint16_t)

   static void OnPlayFinished(DfMp3_PlaySources source, uint16_t track)

               ^

sketch\DFPlayer.h:31:15: note:   candidate expects 2 arguments, 3 provided

[... 106 lines cut ...]
Multiple libraries were found for "SoftwareSerial.h"

 Used: C:\Users\Amfi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\SoftwareSerial

 Not used: C:\Program Files\Arduino\libraries\espsoftwareserial_master

Multiple libraries were found for "ESP8266WiFi.h"

 Used: C:\Users\Amfi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi

 Not used: C:\Program Files\Arduino\libraries\ESP8266WiFi

Multiple libraries were found for "ESP8266HTTPClient.h"

 Used: C:\Users\Amfi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266HTTPClient

 Not used: C:\Program Files\Arduino\libraries\ESP8266HTTPClient

exit status 1

Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

closed 2021-01-13 by Pramfi

miharjo, 2021-02-18:
wah,kasus sama . sudahupdate library terbaru, tapi masih error juga