busel7/JWSMurottal #9: ESP8266wifi.h gak ada filenya

Hammadar77, 2021-01-13:
Arduino: 1.8.5 (Windows 8.1), Board: "Arduino/Genuino Uno"

E:\Kumpulan Program Arduino\JWSMurottal-main\JWSMurottal-main\JWSMurottal\JWSMurottal.ino:1:25: fatal error: ESP8266WiFi.h: No such file or directory

 #include <ESP8266WiFi.h>

                         ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

busel7, 2021-01-14:
Coba update Arduino dan Library sesuai keterangan video

busel7, 2021-02-13:
Mudah-mudahan sudah bisa, case ini kami close. Jika masih ada kendala bisa membuka case ini lagi, terimakasih.

closed 2021-02-13 by busel7