busel7/JWSMurottal #10: Sukses Upload

Pramfi, 2021-01-14:
Assalamualaikum Bang Bonny... Setelah 4 hari nonstop cari errornya karena PENASARAN !!! ...  akhirnya ketemu juga terimakasih bang, buat yg msh menemui error saat compiling coba check di bagian Dfplayer.h  karena cuma disitu penyebabnya, 💯% work asalkan library yg dibutuhkan sudah diinstall semua

busel7, 2021-01-14:
Waalaikumsalam, Alhamdulillah, boleh share lebih detail bagian mana Om? mudah mudahan bisa membantu temen lain yang mengalami masalah yang sama.

abdghofur66, 2021-01-14:
mohon pencerahannya om.. kendala di websockets nya

Hammadar77, 2021-01-21:
wa'alaikumsalam, saya juga kendala di df, mohon di share solusinya

busel7, 2021-02-13:
Mudah-mudahan sudah bisa, case ini kami close. Jika masih ada kendala bisa membuka case ini lagi, terimakasih.

closed 2021-02-13 by busel7

miharjo, 2021-02-18:
> wa'alaikumsalam, saya juga kendala di df, mohon di share solusinya

Sama gan, saya juga kendala, sudah 2 hari update library dll tetep masih eror juga 😭