busel7/JWSMurottal #14: Library Arduino Websockets

arinurcahyo, 2021-02-14:
Dear Mr Bony,

Terima kasih telah sharing JWSMurottal.

Saya sudah install semua libraries sebagaimana tercantum namun masih terjadi error compile : 
**ArduinoWebsockets.h no such file or directory**

Saya cek pada library yang digunakan ArduinoWebsockets.h dari gilmaimon, namun library included pada sketch menggunakan WebSocketsServer.h.

`#include <WebSocketsServer.h>`

Saya coba sesuaikan sketchnya menjadi

`#include <ArduinoWebsockets.h>`

dan sekarang error compile ada pada file WebServer.h dengan pesan error: 
**'WebSocketsServer' does not name a type**

Apakah ada petunjuk?

arinurcahyo, 2021-02-15:
Setelah membaca issues #11, saya kembalikan include pada sketch seperti awal mula yaitu `#include <WebSocketsServer.h>`  dan menginstal library dari websockets dari [https://www.arduino.cc/reference/en/libraries/websockets/](url).
Sketch berhasil dicompile dan diupload ke NodeMCU.
Namun AP tidak mau muncul dan pada terminal monitor muncul keterangan sebagaimana gambar terlampir
[image]

Apakah ada petunjuk? Terima kasih.

busel7, 2021-02-15:
Sip om lupa dicatat kemarin, sudah saya tambah dihalaman depan. Trimakasih feedbacknya.

Untuk error seperti itu karena saat awal belum ada filesettinng yang tersimpan, coba langsung masuk aja ke Setting dan isikan form-form yang kosong nanti akan membuat file-file setting yang diperlukan.

arinurcahyo, 2021-02-15:
Sama-sama pak Bony.
Oh ya btw, setelah aktif, AP dari NodeMCU tidak muncul, jadi saya tidak tahu harus connect kemana :D
Apakah sebelum upload harus ada setting khusus terhadap NodeMCU (menggunakan AT Command misalnya)?

Sebagai info, sebelumnya saya coba upload sketch example untuk webserver dan berjalan normal (AP muncul dan dari PC/ HP bisa konek dan buka contoh webpagenya.

Terima kasih.

busel7, 2021-02-15:
Tidak ada settingan khusus, default kalo tidak ditemukan hotspot akan menjadi aksespoint, bisa dicek disketch bagian werbserver. silahkan diperlajari dan dicoba2. Dan coba update juga arduino ide dan library board untuk espnya mungkin dari situ

arinurcahyo, 2021-02-15:
Baiklah akan saya pelajari dahulu. 
...
Saya download ulang sketch nya, lalu saya compile dan upload ulang, dan alhamdulillah berhasil. 
kabar buruknya: saya tak tahu sebelumnya terdapat kesalahan dimana.

try to figure it out: ternyata SSID dan password untuk AP default terdapat pada webpage.h 
Ok berhasil. I'm in..

Terima kasih Pak Bony, you're rock. :D 

Saya close ya..

closed 2021-02-15 by arinurcahyo