busel7/JWSMurottal #15: Tidak kompatibel dengan ESP-12S (?)

arinurcahyo, 2021-02-15:
Saya menggunakan NodeMCU dari ai-thinker.com dengan chip ESP-12S.
Compile dan upload sketck JWSMurottal berhasil.
Muncul kondisi-kondisi sbb:
1.   Dengan sketch standar, NodeMCU sama sekali tidak dapat koneksi ke wi-fi.
     Setelah saya comment pada file FSConfig.h bagian
```
  if (strlen(configwifi.wifissid) == 0) {
    strlcpy(configwifi.wifissid, "ZAITECH", sizeof(configwifi.wifissid));
  }

if (strlen(configwifi.wifissid) == 0) {             // additional, bawaan BUSEL7 tidak menyertakan password untuk Wifi default
    strlcpy(configwifi.wifipassword, "zait3chhh", sizeof(configwifi.wifipassword));
  }
```
      
dan pada file WebServer.h saya bagian constanta netssid dan netpassword saya explisitkan menjadi:
```
       const char* netssid = "ZAITECH"; //configwifi.wifissid;
       const char* netpassword = "zait3chhh"; //configwifi.wifipassword;
```
NodeMCU dapat tersambung ke wifi, namun selalu putus sambung (terlihat dari Serial Monitor yang selalu mengulang status koneksi dan dicek pada status koneksi pada websetting wifi yang online - offline bergantian terus menerus).

2. dalam kondisi sketch standar, ponsel/ komputer dapat tersambung ke AP NodeMCU

Namun kendalanya adalah Webpage tidak dapat dibuka  baik pada posisi station ataupun AP

Lalu saya belikan NodeMCU baru, kali ini dari LOLIN versi v3 dengan ESP-12E, dengan sketch standar semua masalah langsung hilang. :)

Jadi apakah bisa saya simpulkan JWSMurottal tidak kompatibel dengan ESP-12S?

Ini pertama kalinya menggunakan NodeMCU ESP8266 :D

busel7, 2021-03-13:
Saya belum pernah coba pakai ESP12S, baru coba dengan ESP12E dan ESP12F, kalo pinoutnya sama coba pilih tipe Board lain sebelum Upload di Arduino IDE

busel7, 2022-09-21:
Tidak ada aktifitas lebih lanjut dalam jangka waktu lama.

closed 2022-09-21 by busel7